erik-nelson/blam #1: Test data

sunrainljy, 2016-06-28:
Hi nelson! This is really a cool job. Did you test the BLAM package in Kitti dataset? In addition, could you share the dataset that you used? Thank you very much!

erik-nelson, 2016-06-28:
Hey, thanks!

I didn't test it on KITTI yet, since I didn't see any .bag files to test
with and didn't feel like parsing their single-scan files. I probably will
within the next couple of months. One possible dataset to test with is
here: http://wiki.ros.org/loam_velodyne. Download the "NSH indoor outdoor"
one. Seems to work pretty well :)

Cheers!
Erik

closed 2016-06-28 by erik-nelson

sunrainljy, 2016-06-28:
Hey nelson, Thanks!
I run the "NSH indoor outdoor" dataset. it performs well at the beginning, however, some mistakes happen when the trajectory goes into a small room. The results are shown in the following figure.
[image]
I run the dataset with default parameters. Should I tune the parameters? Do you meet the same problem?
Thank you very much for your consideration.
Best regards!

erik-nelson, 2016-06-28:
Hmm, which launch file are you running? My laptop and 2 lab computers I've tested on run this dataset just fine. If you're replaying the bag file back in real-time (via the online launch file) you could be throttling your processor, and should try the offline launch file instead. Make sure the time scale parameter is set to -1.0 so the messages play back as fast as your CPU can handle them and no faster.

If you're already doing that, I'll take a look tomorrow. You shouldn't have to tune any parameters to get VLP-16 data to work.

sunrainljy, 2016-06-29:
Very thanks! I tried as you said and got a much better results. The results are as follows：
[image]
[image]

There is no mistake in this result. There is only some small drift!
Best regards!