erik-nelson/blam #3: real-time sensor stream

camel007, 2016-07-08:
hi, nelson! I want to test it with my device(velodyne vlp-16), When I type in this command, it show these information and stop working
`roslaunch blam_example test_online.launch 
... logging to /home/nd/.ros/log/a45d32fe-44af-11e6-89b0-408d5c5086b9/roslaunch-nd-Z97X-UD3H-15283.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nd-Z97X-UD3H:51340/

SUMMARY
========

PARAMETERS
 * /blam/blam_slam/check_for_loop_closures: True
 * /blam/blam_slam/filtering/decimate_percentage: 0.9
 * /blam/blam_slam/filtering/grid_filter: True
 * /blam/blam_slam/filtering/grid_res: 0.2
 * /blam/blam_slam/filtering/outlier_filter: False
 * /blam/blam_slam/filtering/outlier_knn: 10
 * /blam/blam_slam/filtering/outlier_std: 1.0
 * /blam/blam_slam/filtering/radius: 0.15
 * /blam/blam_slam/filtering/radius_filter: False
 * /blam/blam_slam/filtering/radius_knn: 3
 * /blam/blam_slam/filtering/random_filter: True
 * /blam/blam_slam/frame_id/base: base
 * /blam/blam_slam/frame_id/fixed: world
 * /blam/blam_slam/frame_id/odometry: odometry
 * /blam/blam_slam/icp/corr_dist: 2.0
 * /blam/blam_slam/icp/iterations: 10
 * /blam/blam_slam/icp/max_rotation: 0.1
 * /blam/blam_slam/icp/max_translation: 0.09
 * /blam/blam_slam/icp/tf_epsilon: 1e-10
 * /blam/blam_slam/icp/transform_thresholding: False
 * /blam/blam_slam/init/orientation/pitch: 0.0
 * /blam/blam_slam/init/orientation/roll: 0.0
 * /blam/blam_slam/init/orientation/yaw: 0.0
 * /blam/blam_slam/init/orientation_sigma/pitch: 0.02
 * /blam/blam_slam/init/orientation_sigma/roll: 0.02
 * /blam/blam_slam/init/orientation_sigma/yaw: 0.02
 * /blam/blam_slam/init/position/x: 0.0
 * /blam/blam_slam/init/position/y: 0.0
 * /blam/blam_slam/init/position/z: 0.0
 * /blam/blam_slam/init/position_sigma/x: 0.1
 * /blam/blam_slam/init/position_sigma/y: 0.1
 * /blam/blam_slam/init/position_sigma/z: 0.1
 * /blam/blam_slam/localization/corr_dist: 1.0
 * /blam/blam_slam/localization/iterations: 10
 * /blam/blam_slam/localization/max_rotation: 0.1
 * /blam/blam_slam/localization/max_translation: 0.05
 * /blam/blam_slam/localization/tf_epsilon: 1e-10
 * /blam/blam_slam/localization/transform_thresholding: False
 * /blam/blam_slam/map/octree_resolution: 0.05
 * /blam/blam_slam/max_tolerable_fitness: 0.15
 * /blam/blam_slam/poses_before_reclosing: 40
 * /blam/blam_slam/proximity_threshold: 1.5
 * /blam/blam_slam/rate/estimate: 20.0
 * /blam/blam_slam/rate/visualization: 0.5
 * /blam/blam_slam/relinearize_skip: 1
 * /blam/blam_slam/relinearize_threshold: 0.01
 * /blam/blam_slam/skip_recent_poses: 40
 * /blam/blam_slam/translation_threshold: 0.5
 * /blam/blam_slam/visualizer/enable_visualization: False
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /use_sim_time: False

NODES
  /blam/
    blam_slam (blam_slam/blam_slam_node)

auto-starting new master
process[master]: started with pid [15295]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a45d32fe-44af-11e6-89b0-408d5c5086b9
process[rosout-1]: started with pid [15308]
started core service [/rosout]
process[blam/blam_slam-2]: started with pid [15322]
[ INFO] [1467943185.583695136]: /blam/blam_slam/BlamSlam: Registering online callbacks.
`
Should I tune some parameters? Thanks for your help! 
Best Regards!

erik-nelson, 2016-07-08:
That's all expected output. After receiving that, it's sitting there waiting for you to give it a sensor stream. Look in the launch file to find the topic that you should publish your VLP-16 point clouds to, and change it if it's not the right topic.

I think by default the VLP-16 publishes to a topic like /velodyne_points or /velodyne, so just set [this line](https://github.com/erik-nelson/blam_example/blob/master/launch/test_online.launch#L13) to whatever your topic is.

camel007, 2016-07-11:
thanks,  Erik! 
why I can't get the result figure as you did?  I installed ros incorrectly or I unfamiliar with ros?

/////////////////////////////////////////////////////////////////////////
There is also no result figure when test it on "NSH indoor-outdoor" dataset
`nd@nd-Z97X-UD3H:~/blam$ roslaunch blam_example test_offline.launch 
... logging to /home/nd/.ros/log/d8afa662-470d-11e6-bb55-408d5c5086b9/roslaunch-nd-Z97X-UD3H-21425.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nd-Z97X-UD3H:32954/

SUMMARY
========

PARAMETERS
 * /blam/blam_slam/check_for_loop_closures: True
 * /blam/blam_slam/filename/bag: /home/nd/blam/nsh...
 * /blam/blam_slam/filtering/decimate_percentage: 0.9
 * /blam/blam_slam/filtering/grid_filter: True
 * /blam/blam_slam/filtering/grid_res: 0.2
 * /blam/blam_slam/filtering/outlier_filter: False
 * /blam/blam_slam/filtering/outlier_knn: 10
 * /blam/blam_slam/filtering/outlier_std: 1.0
 * /blam/blam_slam/filtering/radius: 0.15
 * /blam/blam_slam/filtering/radius_filter: False
 * /blam/blam_slam/filtering/radius_knn: 3
 * /blam/blam_slam/filtering/random_filter: True
 * /blam/blam_slam/frame_id/base: base
 * /blam/blam_slam/frame_id/fixed: world
 * /blam/blam_slam/frame_id/odometry: odometry
 * /blam/blam_slam/icp/corr_dist: 2.0
 * /blam/blam_slam/icp/iterations: 10
 * /blam/blam_slam/icp/max_rotation: 0.1
 * /blam/blam_slam/icp/max_translation: 0.09
 * /blam/blam_slam/icp/tf_epsilon: 1e-10
 * /blam/blam_slam/icp/transform_thresholding: False
 * /blam/blam_slam/init/orientation/pitch: 0.0
 * /blam/blam_slam/init/orientation/roll: 0.0
 * /blam/blam_slam/init/orientation/yaw: 0.0
 * /blam/blam_slam/init/orientation_sigma/pitch: 0.02
 * /blam/blam_slam/init/orientation_sigma/roll: 0.02
 * /blam/blam_slam/init/orientation_sigma/yaw: 0.02
 * /blam/blam_slam/init/position/x: 0.0
 * /blam/blam_slam/init/position/y: 0.0
 * /blam/blam_slam/init/position/z: 0.0
 * /blam/blam_slam/init/position_sigma/x: 0.1
 * /blam/blam_slam/init/position_sigma/y: 0.1
 * /blam/blam_slam/init/position_sigma/z: 0.1
 * /blam/blam_slam/localization/corr_dist: 1.0
 * /blam/blam_slam/localization/iterations: 10
 * /blam/blam_slam/localization/max_rotation: 0.1
 * /blam/blam_slam/localization/max_translation: 0.05
 * /blam/blam_slam/localization/tf_epsilon: 1e-10
 * /blam/blam_slam/localization/transform_thresholding: False
 * /blam/blam_slam/map/octree_resolution: 0.05
 * /blam/blam_slam/max_tolerable_fitness: 0.15
 * /blam/blam_slam/poses_before_reclosing: 40
 * /blam/blam_slam/proximity_threshold: 1.5
 * /blam/blam_slam/rate/estimate: 20.0
 * /blam/blam_slam/rate/visualization: 0.5
 * /blam/blam_slam/relinearize_skip: 1
 * /blam/blam_slam/relinearize_threshold: 0.01
 * /blam/blam_slam/scan_topic: /velodyne_points
 * /blam/blam_slam/skip_recent_poses: 40
 * /blam/blam_slam/time_end: -1.0
 * /blam/blam_slam/time_scale: -1.0
 * /blam/blam_slam/time_start: 0.0
 * /blam/blam_slam/translation_threshold: 0.5
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /use_sim_time: False

NODES
  /blam/
    blam_slam (blam_slam/blam_slam_offline)

auto-starting new master
process[master]: started with pid [21437]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d8afa662-470d-11e6-bb55-408d5c5086b9
process[rosout-1]: started with pid [21450]
started core service [/rosout]
process[blam/blam_slam-2]: started with pid [21464]
[ INFO] [1468203548.405763969]: /blam/blam_slam/BlamSlam: Registering log callbacks.
[ INFO] [1468203548.427442151]: /blam/blam_slam/BlamSlamOffline: Processing the following topics:
	/velodyne_points
[ INFO] [1468203548.436181035]: /blam/blam_slam/BlamSlamOffline: Bag start time    = 1422133388.509928 	 Bag end time    = 1422133476.058435
[ INFO] [1468203548.436230385]: /blam/blam_slam/BlamSlamOffline: Replay start time =         0.000000 	 Replay end time = 4294967296.000000
[ INFO] [1468203548.686504361]: /blam/blam_slam/BlamSlamOffline: Sorting messages.
[ INFO] [1468203548.686723208]: /blam/blam_slam/BlamSlamOffline: Processing messages.
[ INFO] [1468203585.578306241]: /blam/blam_slam/BlamSlamOffline: Finished processing bag file, 237.309657 percent of real-time.

`

Thank you very much!

closed 2016-07-18 by camel007