erik-nelson/blam #6: GTSAM related compile error on Ubuntu 14.04 (using GTSAM 3.2.1)

skohlbr, 2016-08-01:
I get the attached compile output when trying to build. Any ideas for solving are welcome. I'm on a standard 14.04/ROS Indigo system and installed GTSAM 3.2.1 by downloading the archive from the web page as suggested in the GTSAM install instructions. When doing the "make check" I got a failure in the `testWrap` test, but `sudo make install` appeared to work fine.

Start of where things fail (see txt file below for full output):

<pre>Linking CXX executable /home/kohlbrecher/install/blam/internal/devel/lib/point_cloud_mapper/point_cloud_mapper_node
[ 84%] Built target point_cloud_mapper_node
In file included from /usr/local/include/gtsam/geometry/Point3.h:24:0,
                 from /usr/local/include/gtsam/geometry/Pose3.h:29,
                 from /home/kohlbrecher/install/blam/internal/src/laser_loop_closure/include/laser_loop_closure/LaserLoopClosure.h:46,
                 from /home/kohlbrecher/install/blam/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:37:
/usr/local/include/gtsam/base/Matrix.h: In function ‘void gtsam::inplace_QR(MATRIX&)’:
/usr/local/include/gtsam/base/Matrix.h:316:71: error: expected ‘;’ before ‘::’ token
   Eigen::internal::householder_qr_inplace_blocked<MATRIX, HCoeffsType>::run(A, hCoeffs, 48, temp.data());
</pre>

[blam_compile_error.txt](https://github.com/erik-nelson/blam/files/393965/blam_compile_error.txt)

erik-nelson, 2016-08-01:
Hi @skohlbr,

The problem appears to be a versioning issue between the Eigen used by GTSAM and that expected by BLAM. Everything compiles for me when cloning the main GTSAM repository (https://bitbucket.org/gtborg/gtsam/), staying on the ```develop``` branch, and following the standard build instructions:

```
mkdir build
cd build
cmake ..
make check (optional, runs unit tests)
make install
```

Your Ubuntu / ROS versions should be fine. Let me know if that gives you any troubles :)

skohlbr, 2016-08-01:
Thanks for the quick reply! Indeed works after compiling latest GTSAM from source. Also tried the LOAM dataset as described in #1 successfully. 

It might be a good idea to be more specific in the install instructions on how to install GTSAM.

closed 2016-08-01 by skohlbr

erik-nelson, 2016-08-02:
Updated the README. Thanks for the feedback!