erik-nelson/blam #7: Where are these parameters  ?thanks

tpengti, 2016-08-02:
Hey, nelson!
 I dont have found these parameter？
  if (!pu::Get("init/position_sigma/x", sigma_x)) return false;
  if (!pu::Get("init/position_sigma/y", sigma_y)) return false;
  if (!pu::Get("init/position_sigma/z", sigma_z)) return false;
  if (!pu::Get("init/orientation_sigma/roll", sigma_roll)) return false;
  if (!pu::Get("init/orientation_sigma/pitch", sigma_pitch)) return false;
  if (!pu::Get("init/orientation_sigma/yaw", sigma_yaw)) return false;
I didn't find these parameters in "config file"
Thank you very much!

erik-nelson, 2016-08-02:
You will need to run the software with either ```roslaunch blam_example test_online.launch``` or ```roslaunch blam_example test_offline.launch```.

The noted parameters are [here](https://github.com/erik-nelson/blam_example/blob/master/launch/test_offline.launch#L24) and [here](https://github.com/erik-nelson/blam_example/blob/master/launch/test_online.launch#L19), inside of those two launch files.

The online launch file runs the code in real-time, assuming your processor is powerful enough to stream and process LiDAR scans at 10 Hz (the default). Otherwise, you may use the offline launch file to play back a ROS bag file as fast as your computer can handle, or at any desired rate.

tpengti, 2016-08-03:
Thanks for your reply;

tpengti, 2016-08-03:
译文
Hello, ask a question about your use gtsam version？

erik-nelson, 2016-08-03:
I will probably have to freeze in a GTSAM tag pretty soon, but for now use the latest version of the develop branch from https://bitbucket.org/gtborg/gtsam.

tpengti, 2016-08-06:
Thanks !
1:" Pose3 new_odometry = ToGtsam(delta);
  NonlinearFactorGraph new_factor;
  Values new_value;
  new_factor.add(MakeBetweenFactor(new_odometry, ToGtsam(covariance)));"
2:"Pose3 new_odometry = ToGtsam(delta);
  odometry_ = odometry_.compose(new_odometry);
  if (odometry_.translation().norm() < translation_threshold_) {
    return false;
  }"

Why do different code？Which is part of the code you use