erik-nelson/blam #8: Installation

Kailegh, 2016-09-07:
Hi guys, i am trying to install a package (https://github.com/erik-nelson/blam) which says in its instructions:

    This repository contains two ROS workspaces (one internal, one external). The build process is proctored by the update script. To build, first make sure that you do not have any other ROS workspaces in your ROS_PACKAGE_PATH, then clone the repository and from the top directory execute
    
    ./update

My question is, to do that I should delete from my .bashrc the following lines?

    source /opt/ros/indigo/setup.bash
    source ~/catkin_ws/devel/setup.bash

However when I do echo $ROS_PACKAGE_PATH I obtain

/home/dan/catkin_ws/src:/opt/ros/indigo/share:/opt/ros/indigo/stacks

should I also remove the /stacks to install de package?

and Another question, when I install that, to use my normal workspaces, should I source from the terminal? I mean, how can I use my normal workspaces: /home/lsi2/catkin_ws/src:/opt/ros/indigo/share:/opt/ros/indigo/stacks when I install that

Thanks a lot!!!!

Kailegh, 2016-09-07:
well i tried removing those 2 lines from my .bashrc and following your instructions, however it gives me an error ./update: line 13: catkin_make_isolated: order not found
May it be because there is not setup.bash sourced?

Thanks a lot!

erik-nelson, 2016-09-07:
Hi Kailegh. 

You have two lines in your `.bashrc`. The first one:
```
source /opt/ros/indigo/setup.bash
```
tells your system where to find system-wide ROS dependencies, for example, where to find the command `catkin_make_isolated`.

The second one:
```
source ~/catkin_ws/devel/setup.bash
```
tells your system how to find ROS packages that you personally created and put under the `catkin_ws` directory.

To install BLAM, you should keep the first one (hence letting your system know where to find `catkin_make_isolated`), but comment the second one, since your personal packages may conflict with packages within BLAM. The `update` command that I provide will set up two ROS workspaces (similar to your `catkin_ws`) that do not conflict with one another, and will compile them.

In all honesty, it is unlikely that your `catkin_ws` would conflict with any of my packages, but you should comment that line as a precaution.

Kailegh, 2016-09-08:
Thanks a lot, now I have been able to make it work.
I have tried it with different records I have and I have to say that in a moving car(50km/h) it does not work pretty well, in a straight flat road it thinks that u go upwards and makes turns that the car does not do, it also mix the surrounding buildings. I think it does not work very well at high speeds.
However in the tests I have made at lower speeds it works pretty well, even in enviroments where other SLAM software I have used get lost, although it has sometimes some issues (going on a flat trajectory it thinks you change the height).
Congratulations for your work, I am looking forward to new updates.
I will keep testing and giving feedback, and if I can contribute with something useful I will.
Thanks a lot again!!

AshwinMathur, 2016-10-11:
Hey Erik,Kailegh,
Did you make a separate workspace to install BLAM? I keeping getting this error:

/home/hcrobotics/ros_ws/src/blam/internal/src/laser_loop_closure/src/LaserLoopClosure.cc: In member function ‘bool LaserLoopClosure::LoadParameters(const ros::NodeHandle&)’:
/home/hcrobotics/ros_ws/src/blam/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:146:35: error: no matching function for call to ‘gtsam::Pose3::Pose3(gtsam::Rot3&, gtsam::Vector3&)’
   Pose3 pose(rotation, translation);
                                   ^
/home/hcrobotics/ros_ws/src/blam/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:146:35: note: candidates are:
In file included from /home/hcrobotics/ros_ws/src/blam/internal/src/laser_loop_closure/include/laser_loop_closure/LaserLoopClosure.h:46:0,
                 from /home/hcrobotics/ros_ws/src/blam/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:37:
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:78:3: note: gtsam::Pose3::Pose3(const Matrix&)
   Pose3(const Matrix &T) :
   ^
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:78:3: note:   candidate expects 1 argument, 2 provided
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:75:12: note: gtsam::Pose3::Pose3(const gtsam::Pose2&)
   explicit Pose3(const Pose2& pose2);
            ^
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:75:12: note:   candidate expects 1 argument, 2 provided
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:70:3: note: gtsam::Pose3::Pose3(const gtsam::Rot3&, const gtsam::Point3&)
   Pose3(const Rot3& R, const Point3& t) :
   ^
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:70:3: note:   no known conversion for argument 2 from ‘gtsam::Vector3 {aka Eigen::Matrix<double, 3, 1>}’ to ‘const gtsam::Point3&’
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:65:3: note: gtsam::Pose3::Pose3(const gtsam::Pose3&)
   Pose3(const Pose3& pose) :
   ^
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:65:3: note:   candidate expects 1 argument, 2 provided
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:61:3: note: gtsam::Pose3::Pose3()
   Pose3() {
   ^
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:61:3: note:   candidate expects 0 arguments, 2 provided
/home/hcrobotics/ros_ws/src/blam/internal/src/laser_loop_closure/src/LaserLoopClosure.cc: In member function ‘gtsam::Pose3 LaserLoopClosure::ToGtsam(const Transform3&) const’:
/home/hcrobotics/ros_ws/src/blam/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:407:20: error: no matching function for call to ‘gtsam::Pose3::Pose3(gtsam::Rot3&, gtsam::Vector3&)’
   return Pose3(r, t);
                    ^
/home/hcrobotics/ros_ws/src/blam/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:407:20: note: candidates are:
In file included from /home/hcrobotics/ros_ws/src/blam/internal/src/laser_loop_closure/include/laser_loop_closure/LaserLoopClosure.h:46:0,
                 from /home/hcrobotics/ros_ws/src/blam/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:37:
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:78:3: note: gtsam::Pose3::Pose3(const Matrix&)
   Pose3(const Matrix &T) :
   ^
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:78:3: note:   candidate expects 1 argument, 2 provided
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:75:12: note: gtsam::Pose3::Pose3(const gtsam::Pose2&)
   explicit Pose3(const Pose2& pose2);
            ^
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:75:12: note:   candidate expects 1 argument, 2 provided
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:70:3: note: gtsam::Pose3::Pose3(const gtsam::Rot3&, const gtsam::Point3&)
   Pose3(const Rot3& R, const Point3& t) :
   ^
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:70:3: note:   no known conversion for argument 2 from ‘gtsam::Vector3 {aka Eigen::Matrix<double, 3, 1>}’ to ‘const gtsam::Point3&’
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:65:3: note: gtsam::Pose3::Pose3(const gtsam::Pose3&)
   Pose3(const Pose3& pose) :
   ^
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:65:3: note:   candidate expects 1 argument, 2 provided
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:61:3: note: gtsam::Pose3::Pose3()
   Pose3() {
   ^
/home/hcrobotics/catkin_ws/install_isolated/lib/cmake/GTSAM/../../../include/gtsam/geometry/Pose3.h:61:3: note:   candidate expects 0 arguments, 2 provided
/home/hcrobotics/ros_ws/src/blam/internal/src/laser_loop_closure/src/LaserLoopClosure.cc: In member function ‘LaserLoopClosure::Mat66 LaserLoopClosure::ToGu(const shared_ptr&) const’:
[... cut ...]
                   ^
/home/hcrobotics/ros_ws/src/blam/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:428:28: error: ‘gtsam_covariance’ was not declared in this scope
       gtsam_covariance(i, j) = covariance(i, j);
                            ^
/home/hcrobotics/ros_ws/src/blam/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:430:31: error: ‘gtsam_covariance’ was not declared in this scope
   return Gaussian::Covariance(gtsam_covariance);

Will appreciate any help!

Thanks!

Ashwin

alvarezmdavid, 2016-10-25:
Hey Ashwin, Kailegh, Erik

Ashwin, i had the same issue that you're having right now, i think the problem its that you downloaded the gtsam 3.xx.xx, you have to download the 4.00, i solved that problem just doing that, i also made a different workspace to BLAM i called it "/blam_ws" but now i have other issue.... i'll try to solve that by myself but i'm gonna let you know....

Thank you.

AshwinMathur, 2016-11-12:
Hey Alvarez, 
I was able to get it working! Thanks for the help. Hope you got your other issue sorted out. If not let me know, maybe I could help?

eldb2, 2016-11-13:
Hey,

I'm struggling to install the package. I keep getting

` CMake Error: The current CMakeCache.txt directory /home/user/blam-master/internal/build/CMakeCache.txt is different than the directory /home/user/catkin_ws/src/blam-master/internal/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt`

```
Configuring incomplete, errors occurred!
See also "/home/user/blam-master/internal/build/CMakeFiles/CMakeOutput.log".
See also "/home/user/blam-master/internal/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

```

alvarezmdavid, 2016-11-14:
Hey Ashwin, cool, i had an issue about the speed, i'm working with a laptop with a i7 processor, i dont know why but when i play the online test goes in "slow motion", lagging and i really cant map an enviroment, im working in my tesis and i'm kind of stuck right now in matter of SLAM scripts.... this one looks awesome, it works, but not at the speed i need (like in the video that Erik uploaded to youtube).... i don't know if i am doing something wrong or it works the way it should work (slowly)

Thanks for the answer.

nikitaporje, 2016-12-15:
Hi
 I am a complete freshman on ROS and wont to process velodyne data to create point cloud. This is very grate have such project available directly. thanks a lot for this:
I was trying to install the package and create new workspace for that:
after build with ./update its giving me error:

nikita@nikita-Vostro-3800:~/blam_ws/src/blam$ ./update
Creating a ROS workspace for third party packages.
Merge caused no change, no new elements found
Base path: /home/nikita/blam_ws/src/blam/external
Source space: /home/nikita/blam_ws/src/blam/external/src
Build space: /home/nikita/blam_ws/src/blam/external/build_isolated
Devel space: /home/nikita/blam_ws/src/blam/external/devel_isolated
Install space: /home/nikita/blam_ws/src/blam/external/install_isolated
Additional CMake Arguments: -DCMAKE_BUILD_TYPE=Release
No packages found in source space: /home/nikita/blam_ws/src/blam/external/src
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 0 packages in topological order:
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Traceback (most recent call last):
  File "/opt/ros/kinetic/bin/catkin_make_isolated", line 160, in <module>
    main()
  File "/opt/ros/kinetic/bin/catkin_make_isolated", line 156, in main
    override_build_tool_check=opts.override_build_tool_check,
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/catkin/builder.py", line 1050, in build_workspace_isolated
    os.path.join(get_cmake_path(), 'templates', '_setup_util.py.in'),
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/catkin/cmake.py", line 47, in get_cmake_path
    raise RuntimeError('Could not determine catkin cmake path')
RuntimeError: Could not determine catkin cmake path
touch: cannot touch '/home/nikita/blam_ws/src/blam/external/install_isolated/.catkin': No such file or directory
./update: line 16: /home/nikita/blam_ws/src/blam/external/install_isolated/setup.bash: No such file or directory
Creating a ROS workspace for BLAM packages.
Merge caused no change, no new elements found
[blam_example] Updating /home/nikita/blam_ws/src/blam/internal/src/blam_example
[blam_example] Done.
[blam_slam] Updating /home/nikita/blam_ws/src/blam/internal/src/blam_slam
[blam_slam] Done.
[geometry_utils] Updating /home/nikita/blam_ws/src/blam/internal/src/geometry_utils
[geometry_utils] Done.
[laser_loop_closure] Updating /home/nikita/blam_ws/src/blam/internal/src/laser_loop_closure
[laser_loop_closure] Done.
[measurement_synchronizer] Updating /home/nikita/blam_ws/src/blam/internal/src/measurement_synchronizer
[measurement_synchronizer] Done.
[parameter_utils] Updating /home/nikita/blam_ws/src/blam/internal/src/parameter_utils
[parameter_utils] Done.
[point_cloud_filter] Updating /home/nikita/blam_ws/src/blam/internal/src/point_cloud_filter
[point_cloud_filter] Done.
[point_cloud_localization] Updating /home/nikita/blam_ws/src/blam/internal/src/point_cloud_localization
[point_cloud_localization] Done.
[point_cloud_mapper] Updating /home/nikita/blam_ws/src/blam/internal/src/point_cloud_mapper
[point_cloud_mapper] Done.
[point_cloud_odometry] Updating /home/nikita/blam_ws/src/blam/internal/src/point_cloud_odometry
[point_cloud_odometry] Done.
[point_cloud_visualizer] Updating /home/nikita/blam_ws/src/blam/internal/src/point_cloud_visualizer
[point_cloud_visualizer] Done.
[pose_graph_msgs] Updating /home/nikita/blam_ws/src/blam/internal/src/pose_graph_msgs
[pose_graph_msgs] Done.
Base path: /home/nikita/blam_ws/src/blam/internal
Source space: /home/nikita/blam_ws/src/blam/internal/src
Build space: /home/nikita/blam_ws/src/blam/internal/build
Devel space: /home/nikita/blam_ws/src/blam/internal/devel
Install space: /home/nikita/blam_ws/src/blam/internal/install
####
#### Running command: "make cmake_check_build_system" in "/home/nikita/blam_ws/src/blam/internal/build"
####
####
#### Running command: "make -j4 -l4" in "/home/nikita/blam_ws/src/blam/internal/build"
####
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_cpp
[... 26 lines cut ...]
compilation terminated.
compilation terminated.
compilation terminated.
geometry_utils/CMakeFiles/test_base.dir/build.make:62: recipe for target 'geometry_utils/CMakeFiles/test_base.dir/tests/test_base.cc.o' failed
make[2]: *** [geometry_utils/CMakeFiles/test_base.dir/tests/test_base.cc.o] Error 1
CMakeFiles/Makefile2:1541: recipe for target 'geometry_utils/CMakeFiles/test_base.dir/all' failed
make[1]: *** [geometry_utils/CMakeFiles/test_base.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
geometry_utils/CMakeFiles/test_math.dir/build.make:62: recipe for target 'geometry_utils/CMakeFiles/test_math.dir/tests/test_math.cc.o' failed
make[2]: *** [geometry_utils/CMakeFiles/test_math.dir/tests/test_math.cc.o] Error 1
CMakeFiles/Makefile2:1504: recipe for target 'geometry_utils/CMakeFiles/test_math.dir/all' failed
make[1]: *** [geometry_utils/CMakeFiles/test_math.dir/all] Error 2
geometry_utils/CMakeFiles/test_so3error.dir/build.make:62: recipe for target 'geometry_utils/CMakeFiles/test_so3error.dir/tests/test_so3error.cc.o' failed
make[2]: *** [geometry_utils/CMakeFiles/test_so3error.dir/tests/test_so3error.cc.o] Error 1
CMakeFiles/Makefile2:1430: recipe for target 'geometry_utils/CMakeFiles/test_so3error.dir/all' failed
make[1]: *** [geometry_utils/CMakeFiles/test_so3error.dir/all] Error 2
[ 15%] Linking CXX shared library /home/nikita/blam_ws/src/blam/internal/devel/lib/libmeasurement_synchronizer.so
[ 15%] Built target measurement_synchronizer
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

I try to include dependency in CMakeLists.txt but it didn't work.
is there any other dependency that i have to include. 
in my other workspace ros/ros.h is get easily included, I am not getting what the problem is ?

jsgaobiao, 2016-12-20:
I met the same problem as you. @nikitaporje 
Do you have any idea of it right now ?

nikitaporje, 2016-12-21:
thank you @jsgaobiao for reply. :)

TixiaoShan, 2017-02-22:
@alvarezmdavid, have you solved the "slow motion" problem? I am having it also.

ryokochang, 2017-03-18:
Hi, guys

I have already installed ROS and GTSAM, but when I try to run the ./update, I get this error.
Does anyone know how to fix this?

[  0%] [  0%] Built target sensor_msgs_generate_messages_lisp
Built target std_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] [  0%] Built target std_msgs_generate_messages_cpp
Built target sensor_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] [  0%] [  0%] Built target _pose_graph_msgs_generate_messages_check_deps_PoseGraph
Built target _pose_graph_msgs_generate_messages_check_deps_PoseGraphNode
Built target _pose_graph_msgs_generate_messages_check_deps_KeyedScan
[  3%] [  6%] Built target test_base
Built target test_equals
[  9%] [  9%] Built target _pose_graph_msgs_generate_messages_check_deps_PoseGraphEdge
Built target test_math
[ 12%] Built target test_so3error
[ 15%] [ 18%] Built target measurement_synchronizer
Built target point_cloud_mapper
[ 31%] Built target pose_graph_msgs_generate_messages_lisp
[ 46%] Built target pose_graph_msgs_generate_messages_py
[ 50%] Built target point_cloud_filter
[ 53%] Built target point_cloud_localization
[ 65%] Built target pose_graph_msgs_generate_messages_cpp
[ 71%] [ 71%] Built target point_cloud_odometry
Built target point_cloud_visualizer
[ 71%] Built target pose_graph_msgs_generate_messages
[ 75%] Built target point_cloud_filter_node
[ 78%] Built target point_cloud_mapper_node
[ 81%] [ 84%] Built target point_cloud_localization_node
[ 87%] Built target point_cloud_odometry_node
Building CXX object laser_loop_closure/CMakeFiles/laser_loop_closure.dir/src/LaserLoopClosure.cc.o
[ 90%] Built target point_cloud_visualizer_node
In file included from /usr/local/include/gtsam/geometry/Point3.h:24:0,
                 from /usr/local/include/gtsam/geometry/Pose3.h:29,
                 from /home/chang/blam/internal/src/laser_loop_closure/include/laser_loop_closure/LaserLoopClosure.h:46,
                 from /home/chang/blam/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:37:
/usr/local/include/gtsam/base/Matrix.h: In function ‘void gtsam::inplace_QR(MATRIX&)’:
/usr/local/include/gtsam/base/Matrix.h:316:71: error: expected ‘;’ before ‘::’ token
   Eigen::internal::householder_qr_inplace_blocked<MATRIX, HCoeffsType>::run(A, hCoeffs, 48, temp.data());
                                                                       ^
In file included from /usr/local/include/gtsam/geometry/Pose3.h:30:0,
                 from /home/chang/blam/internal/src/laser_loop_closure/include/laser_loop_closure/LaserLoopClosure.h:46,
                 from /home/chang/blam/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:37:
/usr/local/include/gtsam/geometry/Rot3.h: In static member function ‘static gtsam::Rot3 gtsam::Rot3::rodriguez(double, double, double)’:
/usr/local/include/gtsam/geometry/Rot3.h:197:51: error: no matching function for call to ‘gtsam::Rot3::rodriguez(Eigen::CommaInitializer<Eigen::Matrix<double, -1, 1> >&)’
       { return rodriguez((Vector(3) << wx, wy, wz));}
                                                   ^
/usr/local/include/gtsam/geometry/Rot3.h:197:51: note: candidates are:
/usr/local/include/gtsam/geometry/Rot3.h:164:17: note: static gtsam::Rot3 gtsam::Rot3::rodriguez(const Vector&, double)
     static Rot3 rodriguez(const Vector& w, double theta);
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:164:17: note:   candidate expects 2 arguments, 1 provided
/usr/local/include/gtsam/geometry/Rot3.h:172:17: note: static gtsam::Rot3 gtsam::Rot3::rodriguez(const gtsam::Point3&, double)
     static Rot3 rodriguez(const Point3& w, double theta);
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:172:17: note:   candidate expects 2 arguments, 1 provided
/usr/local/include/gtsam/geometry/Rot3.h:180:17: note: static gtsam::Rot3 gtsam::Rot3::rodriguez(const gtsam::Unit3&, double)
     static Rot3 rodriguez(const Unit3& w, double theta);
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:180:17: note:   candidate expects 2 arguments, 1 provided
/usr/local/include/gtsam/geometry/Rot3.h:187:17: note: static gtsam::Rot3 gtsam::Rot3::rodriguez(const Vector&)
     static Rot3 rodriguez(const Vector& v);
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:187:17: note:   no known conversion for argument 1 from ‘Eigen::CommaInitializer<Eigen::Matrix<double, -1, 1> >’ to ‘const Vector& {aka const Eigen::Matrix<double, -1, 1>&}’
[... 99 lines cut ...]
/opt/ros/indigo/include/ros/console.h:342:176: warning: format ‘%lu’ expects argument of type ‘long unsigned int’, but argument 9 has type ‘gtsam::Key {aka unsigned int}’ [-Wformat=]
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                                ^
/opt/ros/indigo/include/ros/console.h:345:5: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER’
     ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER(0, __VA_ARGS__)
     ^
[... cut ...]
/home/chang/blam/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:676:9: note: in expansion of macro ‘ROS_WARN’
         ROS_WARN("%s: Couldn't find timestamp for key %lu", name_.c_str(),
         ^
make[2]: *** [laser_loop_closure/CMakeFiles/laser_loop_closure.dir/src/LaserLoopClosure.cc.o] Error 1
make[1]: *** [laser_loop_closure/CMakeFiles/laser_loop_closure.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

There are also no nodes in my package when I run roscore.
I think the system tries to find the blam_slam_node, but I don't have such file either.

JohnMMeyer, 2017-03-21:
@nikitaporaje and jsgaobiao,
  I had a similar issue as you nikitaporaje and I had to use the solution given by jsgaobiao AND I had to add the following lines to the package.xml file within the package that was throwing the error. 
These were:

_<build_depend>roscpp</build_depend>
<run_depend>roscpp</run_depend>_

JohnMMeyer, 2017-03-21:
Sorry should have mentioned I am running kinetic on 16.04

nikitaporje, 2017-03-27:
@JohnMMeyer Did you edit your CMake file???

JohnMMeyer, 2017-03-27:
Hello @nikitaporje ,
 Yes I did have to modify the CMake file in the package that was giving me trouble.  I added the following:

find_package(catkin REQUIRED COMPONENTS roscpp)
include_directories(include ${catkin_INCLUDE_DIRS})

Let me know if you need any more help,
Take care

volkandre, 2017-09-26:
I created a pull request out of your instructions, see: https://github.com/erik-nelson/geometry_utils/pull/1

daleyadrichem, 2018-01-12:
Anybody who solved the problem of @ryokochang ?

KevinRiehl, 2018-02-06:
Hi guys, I think I solved the problem of ryokochang.
But it is a long story:

1) I am using Ubuntu 16.04.03 and ROS kinetic.
2) I am using Boost Library 1.58
3) I am using a modified form of GTSAM 3.2.1 that is
    compatible with Boost 1.58 and Ubuntu 16.04.03 that
    can be found here https://github.com/indoor-reality/libgtsam
    I had a lot of trouble with libgtsam until I used this
4) I had the same errors as you had. First I had to edit the geometry_utils
    files "CMakeLists.txt" and "package.xml" as you described in this thread before.
5) Then I modified the Code of Blam. Just use this code in the "laser_loop_closure"
     package:
     LaserLoopClosure.cc
`/*
 * Copyright (c) 2016, The Regents of the University of California (Regents).
 * All rights reserved.
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions are
 * met:
 *
 *    1. Redistributions of source code must retain the above copyright
 *       notice, this list of conditions and the following disclaimer.
 *
 *    2. Redistributions in binary form must reproduce the above
 *       copyright notice, this list of conditions and the following
 *       disclaimer in the documentation and/or other materials provided
 *       with the distribution.
 *
 *    3. Neither the name of the copyright holder nor the names of its
 *       contributors may be used to endorse or promote products derived
 *       from this software without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS AS IS
 * AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
 * IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
 * ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE
 * LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
 * CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
 * SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
 * INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
 * CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
 * ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
 * POSSIBILITY OF SUCH DAMAGE.
 *
 * Please contact the author(s) of this library if you have any questions.
 * Author: Erik Nelson            ( eanelson@eecs.berkeley.edu )
 */

#include <laser_loop_closure/LaserLoopClosure.h>

#include <geometry_utils/GeometryUtilsROS.h>
#include <parameter_utils/ParameterUtils.h>
#include <pose_graph_msgs/KeyedScan.h>
#include <pose_graph_msgs/PoseGraph.h>
#include <std_msgs/Empty.h>
#include <visualization_msgs/Marker.h>

#include <pcl/registration/gicp.h>
#include <pcl_conversions/pcl_conversions.h>

namespace gu = geometry_utils;
namespace gr = gu::ros;
namespace pu = parameter_utils;

using gtsam::BetweenFactor;
using gtsam::ISAM2;
using gtsam::ISAM2Params;
using gtsam::NonlinearFactorGraph;
[... 808 lines cut ...]
  ros::Publisher loop_edge_pub_;
  ros::Publisher graph_node_pub_;
  ros::Publisher keyframe_node_pub_;
  ros::Publisher closure_area_pub_;
  ros::Publisher scan1_pub_;
  ros::Publisher scan2_pub_;

  // Pose graph publishers.
  ros::Publisher pose_graph_pub_;
  ros::Publisher keyed_scan_pub_;
  ros::Publisher loop_closure_notifier_pub_;

  typedef std::pair<unsigned int, unsigned int> Edge;
  std::vector<Edge> odometry_edges_;
  std::vector<Edge> loop_edges_;

  // For filtering laser scans prior to ICP.
  PointCloudFilter filter_;
};

#endif`
This should do the trick. Now it is compilable.

I hope I could help you guys.
Sincerely yours, Kevin Riehl