erik-nelson/blam #10: Can not start the rviz

carl1218, 2016-10-03:
hi，
 I try to run the offline mode launch.but it can not start the rviz.
 What should I to set the launch to start the rviz？
 Thank you

started roslaunch server http://ubuntu:50510/

SUMMARY
========

PARAMETERS
 * /blam/blam_slam/check_for_loop_closures: True
 * /blam/blam_slam/filename/bag: /home/robot/Des...
 * /blam/blam_slam/filtering/decimate_percentage: 0.9
 * /blam/blam_slam/filtering/grid_filter: True
 * /blam/blam_slam/filtering/grid_res: 0.2
 * /blam/blam_slam/filtering/outlier_filter: False
 * /blam/blam_slam/filtering/outlier_knn: 10
 * /blam/blam_slam/filtering/outlier_std: 1.0
 * /blam/blam_slam/filtering/radius: 0.15
 * /blam/blam_slam/filtering/radius_filter: False
 * /blam/blam_slam/filtering/radius_knn: 3
 * /blam/blam_slam/filtering/random_filter: True
 * /blam/blam_slam/frame_id/base: base
 * /blam/blam_slam/frame_id/fixed: world
 * /blam/blam_slam/frame_id/odometry: odometry
 * /blam/blam_slam/icp/corr_dist: 2.0
 * /blam/blam_slam/icp/iterations: 10
 * /blam/blam_slam/icp/max_rotation: 0.1
 * /blam/blam_slam/icp/max_translation: 0.09
 * /blam/blam_slam/icp/tf_epsilon: 1e-10
 * /blam/blam_slam/icp/transform_thresholding: False
 * /blam/blam_slam/init/orientation/pitch: 0.0
 * /blam/blam_slam/init/orientation/roll: 0.0
 * /blam/blam_slam/init/orientation/yaw: 0.0
 * /blam/blam_slam/init/orientation_sigma/pitch: 0.02
 * /blam/blam_slam/init/orientation_sigma/roll: 0.02
 * /blam/blam_slam/init/orientation_sigma/yaw: 0.02
 * /blam/blam_slam/init/position/x: 0.0
 * /blam/blam_slam/init/position/y: 0.0
 * /blam/blam_slam/init/position/z: 0.0
 * /blam/blam_slam/init/position_sigma/x: 0.1
 * /blam/blam_slam/init/position_sigma/y: 0.1
 * /blam/blam_slam/init/position_sigma/z: 0.1
 * /blam/blam_slam/localization/corr_dist: 1.0
 * /blam/blam_slam/localization/iterations: 10
 * /blam/blam_slam/localization/max_rotation: 0.1
 * /blam/blam_slam/localization/max_translation: 0.05
 * /blam/blam_slam/localization/tf_epsilon: 1e-10
 * /blam/blam_slam/localization/transform_thresholding: False
 * /blam/blam_slam/map/octree_resolution: 0.05
 * /blam/blam_slam/max_tolerable_fitness: 0.15
 * /blam/blam_slam/poses_before_reclosing: 40
 * /blam/blam_slam/proximity_threshold: 1.5
 * /blam/blam_slam/rate/estimate: 20.0
 * /blam/blam_slam/rate/visualization: 0.5
 * /blam/blam_slam/relinearize_skip: 1
 * /blam/blam_slam/relinearize_threshold: 0.01
 * /blam/blam_slam/scan_topic: /velodyne_points
 * /blam/blam_slam/skip_recent_poses: 40
 * /blam/blam_slam/time_end: -1.0
 * /blam/blam_slam/time_scale: -1.0
 * /blam/blam_slam/time_start: 0.0
 * /blam/blam_slam/translation_threshold: 0.5
 * /rosdistro: indigo
 * /rosversion: 1.11.13
 * /use_sim_time: False

NODES
  /blam/
    blam_slam (blam_slam/blam_slam_offline)

auto-starting new master
process[master]: started with pid [12987]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e42a86e0-892c-11e6-9202-000c29211e88
process[rosout-1]: started with pid [13000]
started core service [/rosout]
process[blam/blam_slam-2]: started with pid [13003]
[ INFO] [1475473659.958945661]: /blam/blam_slam/BlamSlam: Registering log callbacks.
[ INFO] [1475473659.973991522]: /blam/blam_slam/BlamSlamOffline: Processing the following topics:
	/velodyne_points
[ INFO] [1475473668.156287085]: /blam/blam_slam/BlamSlamOffline: Bag start time    = 1422133388.509928 	 Bag end time    = 1422133476.058435
[ INFO] [1475473668.156638780]: /blam/blam_slam/BlamSlamOffline: Replay start time =         0.000000 	 Replay end time = 4294967296.000000
[ INFO] [1475473676.272541490]: /blam/blam_slam/BlamSlamOffline: Sorting messages.
[ INFO] [1475473676.272954454]: /blam/blam_slam/BlamSlamOffline: Processing messages.
[ INFO] [1475473716.824235460]: /blam/blam_slam/BlamSlamOffline: Finished processing bag file, 215.893008 percent of real-time.

JohnMMeyer, 2017-03-27:
Hello @carl1218,
  I always run the command _rviz_ in a second terminal prior to launching BLAM! in a third terminal. Did you get it running?

John

space03, 2019-03-28:
Hi @JohnMMeyer 
I have a question about your comment. 
I launch blam offline in 1st terminal.   _roslaunch blam_example test_offline.launch_ 
and in 2nd terminal, I run rviz so I can check my robot is moving.    _rviz rviz_ 
NOW, how can I export my 3D map? I don't know about this. Please help me.

JohnMMeyer, 2019-03-28:
Hello space03,
 Short answer is I do not know. I haven't messed around with BLAM in a
while, my research has moved in a different direction. i can forward your
question to some people who might know. If I hear back I will let you know.

Good Luck

John M Meyer
Doctoral Candidate Mining Engineering
Professional Master of Mineral Exploration
B.Sc. Geology

Department of Mining Engineering
Colorado School of Mines
1600 Illinois St.
Golden, Colorado. 80401