erik-nelson/blam #11: Installing error at laser_loop_closure

Shokrzadeh, 2016-10-21:
Hi, I installed the GTSAM, I am on ubuntu 14.04, ROS Indigo.
Unzipped the blam-master.
Ran ./update
It installed until 90% and gave this error:
```
make[2]: *** [laser_loop_closure/CMakeFiles/laser_loop_closure.dir/src/LaserLoopClosure.cc.o] Error 1
make[1]: *** [laser_loop_closure/CMakeFiles/laser_loop_closure.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

Also in my .bashrc file only ROS related entry is
`source /opt/ros/indigo/setup.bash`
and I believe that is necessary so I don't know how else to clean my ROS_PACKAGE_PATH or to ensure that there is nothing else there.

Any help on why this happens would be highly appreciated, thanks.

ryokochang, 2017-03-19:
i having the same problem ... any idea to solve this??

JohnMMeyer, 2017-03-21:
Hi all,
 I had the following error at laser_loop_closure, my error involved GTSAM

_CMake Error at laser_loop_closure/CMakeLists.txt:9 (find_package):

  By not providing "FindGTSAM.cmake" in CMAKE_MODULE_PATH this project has

  asked CMake to find a package configuration file provided by "GTSAM", but

  CMake did not find one.

  Could not find a package configuration file provided by "GTSAM" with any of

  the following names:

    GTSAMConfig.cmake

    gtsam-config.cmake

  Add the installation prefix of "GTSAM" to CMAKE_PREFIX_PATH or set

  "GTSAM_DIR" to a directory containing one of the above files.  If "GTSAM"

  provides a separate development package or SDK, be sure it has been

  installed.

-- Configuring incomplete, errors occurred!

See also "/home/johnmeyer/blam_ws/src/blam/internal/build/CMakeFiles/CMakeOutput.log".

See also "/home/johnmeyer/blam_ws/src/blam/internal/build/CMakeFiles/CMakeError.log".

Invoking "cmake" failed_

Not sure if that is exactly what you are getting. If it is I solved it by installing GTSAM with sudo commands. This allowed GTSAM to create a required directory in the /usr  directory.

So I used 

sudo cmake ..
sudo make check
sudo make install

Then it worked fine.

Java-YHB, 2017-07-07:
Hi @JohnMMeyer 
I also meet the problem.Even though I try installing GTSAM with sudo commands,the same errors occur.I guess there may be some thing wrong in the gtsam. Could you tell me the version of gtsam that you use to compile or other solution?
Thanks a lot.

In file included from /usr/local/include/gtsam/geometry/Point3.h:24:0,
                 from /usr/local/include/gtsam/geometry/Pose3.h:29,
                 from /home/ye/blam_ws/blam-master/internal/src/laser_loop_closure/include/laser_loop_closure/LaserLoopClosure.h:46,
                 from /home/ye/blam_ws/blam-master/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:37:
/usr/local/include/gtsam/base/Matrix.h: In function ‘void gtsam::inplace_QR(MATRIX&)’:
/usr/local/include/gtsam/base/Matrix.h:313:71: error: expected ‘;’ before ‘::’ token
   Eigen::internal::householder_qr_inplace_blocked<MATRIX, HCoeffsType>::run(A, hCoeffs, 48, temp.data());
                                                                       ^
In file included from /usr/local/include/gtsam/geometry/Pose3.h:30:0,
                 from /home/ye/blam_ws/blam-master/internal/src/laser_loop_closure/include/laser_loop_closure/LaserLoopClosure.h:46,
                 from /home/ye/blam_ws/blam-master/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:37:
/usr/local/include/gtsam/geometry/Rot3.h: In static member function ‘static gtsam::Rot3 gtsam::Rot3::rodriguez(double, double, double)’:
/usr/local/include/gtsam/geometry/Rot3.h:197:51: error: no matching function for call to ‘gtsam::Rot3::rodriguez(Eigen::CommaInitializer<Eigen::Matrix<double, -1, 1> >&)’
       { return rodriguez((Vector(3) << wx, wy, wz));}
                                                   ^
/usr/local/include/gtsam/geometry/Rot3.h:197:51: note: candidates are:
/usr/local/include/gtsam/geometry/Rot3.h:164:17: note: static gtsam::Rot3 gtsam::Rot3::rodriguez(const Vector&, double)
     static Rot3 rodriguez(const Vector& w, double theta);
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:164:17: note:   candidate expects 2 arguments, 1 provided
/usr/local/include/gtsam/geometry/Rot3.h:172:17: note: static gtsam::Rot3 gtsam::Rot3::rodriguez(const gtsam::Point3&, double)
     static Rot3 rodriguez(const Point3& w, double theta);
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:172:17: note:   candidate expects 2 arguments, 1 provided
/usr/local/include/gtsam/geometry/Rot3.h:180:17: note: static gtsam::Rot3 gtsam::Rot3::rodriguez(const gtsam::Unit3&, double)
     static Rot3 rodriguez(const Unit3& w, double theta);
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:180:17: note:   candidate expects 2 arguments, 1 provided
/usr/local/include/gtsam/geometry/Rot3.h:187:17: note: static gtsam::Rot3 gtsam::Rot3::rodriguez(const Vector&)
     static Rot3 rodriguez(const Vector& v);
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:187:17: note:   no known conversion for argument 1 from ‘Eigen::CommaInitializer<Eigen::Matrix<double, -1, 1> >’ to ‘const Vector& {aka const Eigen::Matrix<double, -1, 1>&}’
/usr/local/include/gtsam/geometry/Rot3.h:196:17: note: static gtsam::Rot3 gtsam::Rot3::rodriguez(double, double, double)
     static Rot3 rodriguez(double wx, double wy, double wz)
                 ^
/usr/local/include/gtsam/geometry/Rot3.h:196:17: note:   candidate expects 3 arguments, 1 provided
In file included from /home/ye/blam_ws/blam-master/internal/src/laser_loop_closure/include/laser_loop_closure/LaserLoopClosure.h:46:0,
                 from /home/ye/blam_ws/blam-master/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:37:
/usr/local/include/gtsam/geometry/Pose3.h: In static member function ‘static gtsam::Matrix gtsam::Pose3::wedge(double, double, double, double, double, double)’:
/usr/local/include/gtsam/geometry/Pose3.h:226:28: error: could not convert ‘(&(&(&(&(&(&(&(&(&(&(&(&(&(& Eigen::DenseBase<Derived>::operator<<(const Scalar&) [with Derived = Eigen::Matrix<double, -1, -1>; Eigen::DenseBase<Derived>::Scalar = double]((* &0.0)).Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((* &(- wz))))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((*(const Scalar*)(& wy))))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((*(const Scalar*)(& vx))))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((*(const Scalar*)(& wz))))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((* &0.0)))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((* &(- wx))))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((*(const Scalar*)(& vy))))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((* &(- wy))))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((*(const Scalar*)(& wx))))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((* &0.0)))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((*(const Scalar*)(& vz))))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((* &0.0)))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((* &0.0)))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix<double, -1, -1> >((* &0.0)))->Eigen::CommaInitializer<MatrixType>::operator,<Eigen::Matrix< [... cut ...]
       gtsam_covariance(i, j) = covariance(i, j);
                            ^
/home/ye/blam_ws/blam-master/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:430:31: error: ‘gtsam_covariance’ was not declared in this scope
   return Gaussian::Covariance(gtsam_covariance);
                               ^
make[2]: *** [laser_loop_closure/CMakeFiles/laser_loop_closure.dir/src/LaserLoopClosure.cc.o] Error 1
make[1]: *** [laser_loop_closure/CMakeFiles/laser_loop_closure.dir/all] Error 2
make: *** [all] Error 2

einsnull, 2017-11-12:
@Java-YHB Did you solve the problem?  It also occurs to me.

KevinRiehl, 2018-02-07:
I think I solved your problems guys. Have a look on the "installation" issue:
https://github.com/erik-nelson/blam/issues/8

I hope I could help you
Sincerely yours, Kevin Riehl

whuzxy, 2018-05-25:
It is the problem of the vision of GTSAM. I got the same error in gtsam3.2.1.There are no errors when I install 4.0.

closed 2021-04-05 by Shokrzadeh