erik-nelson/blam #15: Data visualization

nikitaporje, 2017-01-19:
can we view the data in cloud compare ? how to export the data just for visualization?

JohnMMeyer, 2017-03-21:
Hello nikitaporje,
 I have the same question. What I have done so far is write the bag file as a PCD file which is then importable to CloudCompare using pcl_ros (http://wiki.ros.org/pcl_ros) as follows:

_rosrun pcl_ros bag_to_pcd <input_file.bag> <topic> <output_directory>_

Use that command in the directory containing your bag file, it will write the PCD files to an output directory

The problem is that the PCD files are not registered.  When one imports them into CloudComapre, they are all in the LiDAR frame of reference when we need them in the world frame. We need to somehow record and import the registered point clouds.

Luoada, 2017-04-02:
@JohnMMeyer @nikitaporje @erik-nelson 
Hello guys,
Has anyone tried building this project for realtime SLAM using VLP16?
Actually when I ran this test_online.launch file, I could not see anything ar rviz window.

nikitaporje, 2017-04-03:
@JohnMMeyer Try to collect the PCD file by using 
`rosrun pcl_ros pointcloud_to_pcd input:=/blam/blamSlam/Octree_map_updates`
then you get multiple pcd file with accurate positioning

nikitaporje, 2017-04-03:
@Luoada I did not try to run the project with live sensor attached. But i play the bag file with VLP16 data and run the test_online.launch it work fine for me.

Luoada, 2017-04-03:
@nikitaporje Hi Nikitaporje,
Thanks for your reply. So, if I am not mistaken did you use your own VLP-16 bag file or the same files among author have provided. Many thanks:)
Best,
Ada

nikitaporje, 2017-04-03:
@Luoada I also tried with the velodyne VLP16 raw data in pcap format 
firstly i convert pcap file in bag file then use it

Luoada, 2017-04-03:
@nikitaporje 
Thanks a lot:) I will try it out soon.
if you don't mind could you please text at luoada88@gmail.com or share your email id. Actually, I have few queries and want to get reviewed few results based on VLP16 data.
Your valuable feedback is very much appreciated.

Best regards,
Ada

JohnMMeyer, 2017-04-03:
Hi @nikitaporje,
Thanks for getting back to me. So I was using `rosrun pcl_ros pointcloud_to_pcd input:=/blam/blamSlam/Octree_map`   which was giving me a registered point cloud but it did not have a lot of data. Your suggestion really helped as I am now getting 1,800,000 points as opposed to 100,000 when processing the same bagfile. For what it is worth, I took the multiple .pcd files that your command , `rosrun pcl_ros pointcloud_to_pcd input:=/blam/blamSlam/Octree_map_updates`  generated, loaded them inot CloudCompare, then used the merge function on CloudCompare to generate one large .pcd file. So thanks, that was very helpful.

ranxiaoustc, 2017-06-12:
@nikitaporje @JohnMMeyer @Luoada @erik-nelson Did you get the real-time 3D map with vlp-16? I mean we do not convert the pcap ifle. 
Thanks for your reply.

nikitaporje, 2017-06-13:
@JohnMMeyer @erik-nelson  do you have any real time data available from your sensor with GPS positioning data.Can you please pin me on my email account nporje@gmail.com i have some quires related to that.
thank you for help

nikitaporje, 2017-06-13:
@ranxiaoustc I haven't tested the real time 3D mapping using VLP-16. I have tested only off-line data.

JohnMMeyer, 2017-06-14:
Hello all,
  Everything I have collected has been underground so I do not have any GPS
data, sorry about that.

John

John M Meyer
Doctoral Candidate Mining Engineering
Professional Master of Mineral Exploration
B.Sc. Geology

Department of Mining Engineering
Colorado School of Mines
1600 Illinois St.
Golden, Colorado. 80401