erik-nelson/blam #18: Increasing point density in octree_map

JohnMMeyer, 2017-03-31:
Hello,
 I have been working with BLAM! for a bit now, it is working well for me- Erik thank you so much for contributing this software.
I am using BLAM! as part of a process of building 3D maps using LiDAR data with 6 DoF. I am using BLAM! to generate a registered point cloud, I then convert the octree_map data into .pcd files, convert those into .las files and then manipulate them. 

What I was wondering was if it was possible to increase the number of points that are outputted with the octree_map node. I am getting good maps but there is not a lot of data points in the cloud that is being output. 

Thanks for any help,
John

mohit-khosti, 2019-08-19:
Hi @JohnMMeyer 
Can you please help me... I have successfully installed BLAM, however I am not sure how to visualize the BLAM data with a running LiDAR sensor :( I am using Ouster LiDAR sensor.
Thank you for your help!

JohnMMeyer, 2019-08-19:
I have used RViz

mohit-khosti, 2019-08-19:
Hi @JohnMMeyer 
Thank you for your response, I have already done that however I see nothing. Does it have something to do with the Topic or is there some specific way to let BLAM communicate with LiDAR sensor?
Thank you!

JohnMMeyer, 2019-08-19:
Hi Mohit,
 I have not messed with this in a while and I am super rusty. I would make
sure your topic is publishing.

John M Meyer
Doctoral Candidate Mining Engineering
Professional Master of Mineral Exploration
B.Sc. Geology

Department of Mining Engineering
Colorado School of Mines
1600 Illinois St.
Golden, Colorado. 80401