erik-nelson/blam #19: speed issue

JohannLac, 2017-05-17:
Hi,
When I launched the online or offline launch file, the mapping is really slow (not real-time at all). I work on Ubuntu 64bits 14.04 with ROS indigo (i7-2670QM 2.20GHz x 8 core, 16Gb ram, GeForce GTX 560M).
Is there any parameter to change, or a computer configuration that make this soft running slow ? I tried several bag or pcap file, including nsh_indoor_outdoor from LOAM, and it is always the same.

Thanks for answers in advance :)

jingxiaoliguang, 2017-07-22:
Hi, @JohannLac . I encountered the same problem. Have you solved it?

JohannLac, 2017-07-24:
Unfortunately no... Sorry !
I tried others SLAM which work for me (like LOAM), so I didn't spend much time trying to make this one work.

taogashi, 2017-07-25:
`pcl::GeneralizedIterativeClosestPoint` contributes a lot...Try `pcl::IterativeClosestPoint` or `pcl::IterativeClosestPointWithNormals`, there's a tradeoff between speed and accuracy.

JohannLac, 2017-07-25:
I'll try ! Thanks :)

jingxiaoliguang, 2017-07-25:
Thanks a lot!☺ I'll try it.

01001HR, 2017-07-25:
Hi, @JohannLac . Is there any other 3D ladir SLAM we can try, i didn't find more except LOAM?

JohannLac, 2017-07-25:
Hi,
I didn't find any other (open source) 3D SLAM either.
I converted my pointCloud into a LaserScan in order to work with 2D SLAM like hector (which works fine), but if you absolutely want a 3D SLAM I only found LOAM :/

01001HR, 2017-07-26:
@JohannLac, thanks.

YuxiHuang001, 2017-08-14:
I think maybe you subscribe the "/blam/blam_slam/octree_map" topic， it publishes all the map,so the map will become larger and larger ,and the system will be more and more slowly.

antithing, 2017-09-26:
Hi all, I am also struggling to get realtime speeds. Did anyone manage to get anything close to 50ms per frame? I am currently at 200+. 

I use organised clouds, but the voxel filtering organizes them...

Choco83, 2017-10-25:
I encountered the same issue, will try what @taogashi suggested. Those of you looking for 3d SLAM can try out cartographer maybe, it's a bit resource intensive and requires an IMU though. But the support is really good and large community is actively using it.