erik-nelson/blam #21: blam_slam_node run error

manomitbal, 2017-06-29:
Hi!

I would like to start by congratulating you on a lovely algorithm. That being said, I am encountering a very peculiar error where the transformation matrices are giving out NaN values whenever I try to launch the blam slam node. The following is the output:

`started roslaunch server http://192.168.1.77:43921/

SUMMARY
========

PARAMETERS
 * /blam/blam_slam/frame_id/base: zed_initial_frame
 * /blam/blam_slam/frame_id/fixed: map
 * /blam/blam_slam/frame_id/odometry: odom
 * /blam/blam_slam/init/orientation/pitch: 0.0
 * /blam/blam_slam/init/orientation/roll: 0.0
 * /blam/blam_slam/init/orientation/yaw: 0.0
 * /blam/blam_slam/init/orientation_sigma/pitch: 0.02
 * /blam/blam_slam/init/orientation_sigma/roll: 0.02
 * /blam/blam_slam/init/orientation_sigma/yaw: 0.02
 * /blam/blam_slam/init/position/x: 0.0
 * /blam/blam_slam/init/position/y: 0.0
 * /blam/blam_slam/init/position/z: 0.0
 * /blam/blam_slam/init/position_sigma/x: 0.1
 * /blam/blam_slam/init/position_sigma/y: 0.1
 * /blam/blam_slam/init/position_sigma/z: 0.1
 * /blam/blam_slam/rate/estimate: 20.0
 * /blam/blam_slam/rate/visualization: 0.5
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /use_sim_time: False

NODES
  /blam/
    blam_slam (blam_slam/blam_slam_node)

ROS_MASTER_URI=http://192.168.1.80:11311

core service [/rosout] found
process[blam/blam_slam-1]: started with pid [4733]
[ INFO] [1498763198.239834200]: /blam/blam_slam/BlamSlam: Registering online callbacks.
blam_slam_node: /build/pcl-6_P28C/pcl-1.7.2/kdtree/include/pcl/kdtree/impl/kdtree_flann.hpp:136: int pcl::KdTreeFLANN<PointT, Dist>::nearestKSearch(const PointT&, int, std::vector<int>&, std::vector<float>&) const [with PointT = pcl::PointXYZ; Dist = flann::L2_Simple<float>]: Assertion `point_representation_->isValid (point) && "Invalid (NaN, Inf) point coordinates given to nearestKSearch!"' failed.
[blam/blam_slam-1] process has died [pid 4733, exit code -6, cmd /home/fx250/blam_ws/devel/lib/blam_slam/blam_slam_node ~pcld:=/zed/point_cloud/cloud_registered __name:=blam_slam __log:=/home/fx250/.ros/log/166c611c-5cfb-11e7-8538-001fc69c3cb2/blam-blam_slam-1.log].
log file: /home/fx250/.ros/log/166c611c-5cfb-11e7-8538-001fc69c3cb2/blam-blam_slam-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

Any help is appreciated! Thanks in advance!

Nickhodem, 2017-07-31:
Hi,
I think you should remove every NaN points in point cloud, before you sent it to blam. 
pcl::PointCloud<pcl::PointXYZ> laserCloudIn;
pcl::PointCloud<pcl::PointXYZ>::Ptr laserCloudWithoutNaN (new pcl::PointCloud<pcl::PointXYZ> ());
std::vector<int> indices;
pcl::removeNaNFromPointCloud(laserCloudIn, *laserCloudWithoutNaN, indices);

tiberium24, 2018-12-08:
These directions are not correct. Can someone say where we should make this conversion and how? Since it is very common to get Nan values, we should fix this.

Disonye, 2019-11-20:
filter里面没有去除nan点，建议加载点云后去除nan点；
if (!points->is_dense)
  {
    points_filtered->is_dense = false;
    std::vector<int> indices;
    pcl::removeNaNFromPointCloud(*points_filtered,*points_filtered, indices);
  }

shrr98, 2020-02-01:
> filter里面没有去除nan点，建议加载点云后去除nan点；
> if (!points->is_dense)
> {
> points_filtered->is_dense = false;
> std::vector indices;
> pcl::removeNaNFromPointCloud(*points_filtered,*points_filtered, indices);
> }

does this really fix the problem? I have tried the solution but the issue is still there.