erik-nelson/blam #26: PCL error?

capaulson, 2017-09-29:
We're trying to run this code and encountered the following error as soon as the LIDAR started streaming data: 

```blam_slam_node: /build/pcl-6_P28C/pcl-1.7.2/kdtree/include/pcl/kdtree_flann.hpp:136: int pcl::KdTreeFLANN<PointT, Dist>::nearestKSearch(const PointT&, int, std::vector<int>&, std::vector<float>&) const [with PointT = pcl::PointXYZ: Dist = flann::L2simple<float>] Assertion 'point_representation_->isValid (point) && "Invalid (NaN, Inf) point coordinates given to nearestKSearch!"' failed.```

Any ideas?

wzy0426, 2018-04-08:
hey, I met the same problam. Did you fix it ?

caylaXu, 2018-07-18:
I have solved the problem

Probably because your point cloud data contains"is_dense:false",It means that your point cloud data contains nan

my solution is edit the “point_cloud_filter.cc”， add the code at the end of the Filter function

  if (!points->is_dense)
  {
    points_filtered->is_dense = false;
    std::vector<int> indices;
    pcl::removeNaNFromPointCloud(*points_filtered,*points_filtered, indices);
  }

shrr98, 2020-02-01:
> I have solved the problem
> 
> Probably because your point cloud data contains"is_dense:false",It means that your point cloud data contains nan
> 
> my solution is edit the “point_cloud_filter.cc”， add the following code at the end of the Filter function
> 
> if (!points->is_dense)
> {
> points_filtered->is_dense = false;
> std::vector indices;
> pcl::removeNaNFromPointCloud(*points_filtered,*points_filtered, indices);
> }

I have tried your solution but it doesn't fix the issue.

nightlygeek12, 2020-10-08:
> I have tried your solution but it doesn't fix the issue.

if (!points->is_dense)
{
points_filtered->is_dense = false;
std::vector<int> indices;
pcl::removeNaNFromPointCloud(*points_filtered,*points_filtered, indices);
}

Try by adding "<int>" iafter vector. 

This solved the issue and I was able to run it.