erik-nelson/blam #29: Update Error

choisangwook, 2018-02-20:
Hello ...!
I am in the environment of ubuntu 16.04.3, kinetic ROS and the following error occurs:
I'd appreciate your help.

sangwook@sangwook-VirtualBox:~/catkin_ws/src$ pwd
/home/sangwook/catkin_ws/src
sangwook@sangwook-VirtualBox:~/catkin_ws/src$ ls -al
합계 320
drwxrwxr-x 12 sangwook sangwook   4096  2월 20 20:07 .
drwxrwxr-x  5 sangwook sangwook   4096  2월 20 19:23 ..
-rw-rw-r--  1 sangwook sangwook  30286  2월 20 19:50 CMakeCache.txt
drwxrwxr-x  9 sangwook sangwook   4096  2월 20 19:51 CMakeFiles
lrwxrwxrwx  1 sangwook sangwook     50  2월 20 19:23 CMakeLists.txt -> /opt/ros/kinetic/share/catkin/cmake/toplevel.cmake
-rw-r--r--  1 sangwook sangwook   3447  2월 20 19:51 CPackConfig.cmake
-rw-r--r--  1 sangwook sangwook   3753  2월 20 19:51 CPackSourceConfig.cmake
-rw-rw-r--  1 sangwook sangwook    392  2월 20 19:50 CTestTestfile.cmake
drwxrwxr-x  3 sangwook sangwook   4096  2월 20 19:51 CppUnitLite
-rw-rw-r--  1 sangwook sangwook   3169  2월 20 19:50 GTSAM-exports.cmake
-rw-rw-r--  1 sangwook sangwook    785  2월 20 19:50 GTSAMConfig.cmake
-rw-rw-r--  1 sangwook sangwook 209662  2월 20 19:51 Makefile
drwxrwxr-x  5 sangwook sangwook   4096  2월 20 20:07 blam
drwxrwxr-x  3 sangwook sangwook   4096  2월 20 19:51 cmake
-rw-rw-r--  1 sangwook sangwook   4788  2월 20 19:50 cmake_install.cmake
drwxrwxr-x  3 sangwook sangwook   4096  2월 20 19:51 examples
drwxrwxr-x 13 sangwook sangwook   4096  2월 20 19:51 gtsam
-rw-rw-r--  1 sangwook sangwook    389  2월 20 19:50 gtsam_extra.cmake
drwxrwxr-x 12 sangwook sangwook   4096  2월 20 19:34 libgtsam
drwxrwxr-x  3 sangwook sangwook   4096  2월 20 19:51 tests
drwxrwxr-x  3 sangwook sangwook   4096  2월 20 19:51 timing
drwxrwxr-x  4 sangwook sangwook   4096  2월 20 19:51 wrap
sangwook@sangwook-VirtualBox:~/catkin_ws/src$ cd blam
sangwook@sangwook-VirtualBox:~/catkin_ws/src/blam$ ./update
Creating a ROS workspace for third party packages.
Merge caused no change, no new elements found
Base path: /home/sangwook/catkin_ws/src/blam/external
Source space: /home/sangwook/catkin_ws/src/blam/external/src
Build space: /home/sangwook/catkin_ws/src/blam/external/build_isolated
Devel space: /home/sangwook/catkin_ws/src/blam/external/devel_isolated
Install space: /home/sangwook/catkin_ws/src/blam/external/install_isolated
Additional CMake Arguments: -DCMAKE_BUILD_TYPE=Release
No packages found in source space: /home/sangwook/catkin_ws/src/blam/external/src
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 0 packages in topological order:
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
touch: '/home/sangwook/catkin_ws/src/blam/external/install_isolated/.catkin'를 touch할 수 없음: 그런 파일이나 디렉터리가 없습니다
./update: 줄 16: /home/sangwook/catkin_ws/src/blam/external/install_isolated/setup.bash: 그런 파일이나 디렉터리가 없습니다
Creating a ROS workspace for BLAM packages.
Merge caused no change, no new elements found
[blam_example] Updating /home/sangwook/catkin_ws/src/blam/internal/src/blam_example
[blam_example] Done.
[blam_slam] Updating /home/sangwook/catkin_ws/src/blam/internal/src/blam_slam
[blam_slam] Done.
[geometry_utils] Updating /home/sangwook/catkin_ws/src/blam/internal/src/geometry_utils
[geometry_utils] Done.
[laser_loop_closure] Updating /home/sangwook/catkin_ws/src/blam/internal/src/laser_loop_closure
[laser_loop_closure] Done.
[measurement_synchronizer] Updating /home/sangwook/catkin_ws/src/blam/internal/src/measurement_synchronizer
[measurement_synchronizer] Done.
[parameter_utils] Updating /home/sangwook/catkin_ws/src/blam/internal/src/parameter_utils
[parameter_utils] Done.
[point_cloud_filter] Updating /home/sangwook/catkin_ws/src/blam/internal/src/point_cloud_filter
[point_cloud_filter] Done.
[point_cloud_localization] Updating /home/sangwook/catkin_ws/src/blam/internal/src/point_cloud_localization
[point_cloud_localization] Done.
[point_cloud_mapper] Updating /home/sangwook/catkin_ws/src/blam/internal/src/point_cloud_mapper
[point_cloud_mapper] Done.
[point_cloud_odometry] Updating /home/sangwook/catkin_ws/src/blam/internal/src/point_cloud_odometry
[point_cloud_odometry] Done.
[point_cloud_visualizer] Updating /home/sangwook/catkin_ws/src/blam/internal/src/point_cloud_visualizer
[point_cloud_visualizer] Done.
[... 119 lines cut ...]
-- looking for PCL_SAMPLE_CONSENSUS
-- looking for PCL_FILTERS
-- looking for PCL_FEATURES
-- looking for PCL_REGISTRATION
-- Found PCL: /usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libboost_atomic.so;/usr/lib/x86_64-linux-gnu/libboost_regex.so;/usr/lib/x86_64-linux-gnu/libpthread.so;optimized;/usr/lib/x86_64-linux-gnu/libpcl_common.so;debug;/usr/lib/x86_64-linux-gnu/libpcl_common.so;optimized;/usr/lib/x86_64-linux-gnu/libpcl_octree.so;debug;/usr/lib/x86_64-linux-gnu/libpcl_octree.so;optimized;/usr/lib/x86_64-linux-gnu/libflann_cpp_s.a;debug;/usr/lib/x86_64-linux-gnu/libflann_cpp_s.a;optimized;/usr/lib/x86_64-linux-gnu/libpcl_kdtree.so;debug;/usr/lib/x86_64-linux-gnu/libpcl_kdtree.so;optimized;/usr/lib/x86_64-linux-gnu/libpcl_search.so;debug;/usr/lib/x86_64-linux-gnu/libpcl_search.so;optimized;/usr/lib/x86_64-linux-gnu/libpcl_sample_consensus.so;debug;/usr/lib/x86_64-linux-gnu/libpcl_sample_consensus.so;optimized;/usr/lib/x86_64-linux-gnu/libpcl_filters.so;debug;/usr/lib/x86_64-linux-gnu/libpcl_filters.so;optimized;/usr/lib/x86_64-linux-gnu/libpcl_features.so;debug;/usr/lib/x86_64-linux-gnu/libpcl_features.so;optimized;/usr/lib/x86_64-linux-gnu/libpcl_registration.so;debug;/usr/lib/x86_64-linux-gnu/libpcl_registration.so;/usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libboost_atomic.so;/usr/lib/x86_64-linux-gnu/libboost_regex.so;/usr/lib/x86_64-linux-gnu/libpthread.so;optimized;/usr/lib/x86_64-linux-gnu/libflann_cpp_s.a;debug;/usr/lib/x86_64-linux-gnu/libflann_cpp_s.a  
CMake Error at laser_loop_closure/CMakeLists.txt:9 (find_package):
  By not providing "FindGTSAM.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "GTSAM", but
  CMake did not find one.

  Could not find a package configuration file provided by "GTSAM" with any of
  the following names:

    GTSAMConfig.cmake
    gtsam-config.cmake

  Add the installation prefix of "GTSAM" to CMAKE_PREFIX_PATH or set
  "GTSAM_DIR" to a directory containing one of the above files.  If "GTSAM"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/sangwook/catkin_ws/src/blam/internal/build/CMakeFiles/CMakeOutput.log".
See also "/home/sangwook/catkin_ws/src/blam/internal/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

TobiMiller, 2018-03-01:
you need to compile GTSAM from source by yourself. look into README file.

choisangwook, 2018-03-07:
Due to problems with the ftp server, the download could not be properly made
It's clear now
Thank you