erik-nelson/blam #34: Installation - "update" script - Fatal error: ros/ros.h: no such file or dir

wangyiwei2015, 2018-08-20:
No error occurred before "running command 'make -j4 -l4' ..."

mfassler, 2018-10-23:
@graberj I created this patch.  These are the changes that I had to make to get it to compile and run.  

If you cd into geometry_utils, you can run this:
```
patch -p1 </path/to/geometry_utils.patch.txt
```
... and it should make the changes to the 4 files.  

(if you look into the patch, you can see that I added 2 lines to package.xml, I added 1 line to test_base.cc and deleted 1 line from test_base.cc, and I deleted 1 line from each of test_math.cc and test_so3error.cc.   But the patch command above should do this all for you automatically.)

ImtiazUlHassan, 2022-09-09:
Go to 

> blam/internal/src/geometry_utils

Add the following two lines to package.xml

```
<build_depend>roscpp</build_depend>
<run_depend>roscpp</run_depend>
```

Add the following two lines to CMakeList.txt

```
find_package(catkin REQUIRED COMPONENTS roscpp)
include_directories(include ${catkin_INCLUDE_DIRS})
```

Reference :[https://blog.katastros.com/a?ID=01050-a0697a84-317f-423f-b16d-cd90c455da97](url)

Gypsop, 2023-11-24:
> Go to
>
>
> Add the following two lines to package.xml
>
> ```
> <build_depend>roscpp</build_depend>
> <run_depend>roscpp</run_depend>
> ```
>
> Add the following two lines to CMakeList.txt
>
> ```
> find_package(catkin [...]

This worked for me. Thanks!