erik-nelson/blam #35: blam-external.rosinstall is empty

louis960126, 2018-09-11:
Hi erik, 
    I tried to merge some data from blam-external.rosinstall, but it was empty. Is it normal?

    and I did catkin_make but i met some problems 

-- Build files have been written to: /home/vincent/catkin_ws/blam/internal/build
####
#### Running command: "make -j1 -l1" in "/home/vincent/catkin_ws/blam/internal/build"
####
Scanning dependencies of target test_so3error
[  1%] Building CXX object geometry_utils/CMakeFiles/test_so3error.dir/tests/test_so3error.cc.o
/home/vincent/catkin_ws/blam/internal/src/geometry_utils/tests/test_so3error.cc:5:21: fatal error: ros/ros.h: No such file or directory
compilation terminated.
geometry_utils/CMakeFiles/test_so3error.dir/build.make:62: recipe for target 'geometry_utils/CMakeFiles/test_so3error.dir/tests/test_so3error.cc.o' failed
make[2]: *** [geometry_utils/CMakeFiles/test_so3error.dir/tests/test_so3error.cc.o] Error 1
CMakeFiles/Makefile2:468: recipe for target 'geometry_utils/CMakeFiles/test_so3error.dir/all' failed
make[1]: *** [geometry_utils/CMakeFiles/test_so3error.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1 -l1" failed

please help me solve the problem thanks!

aaravrav142, 2018-11-14:
> Hi, you could replace the two uploaded files in ${your_catkin_ws}/internal/src/geometry_utils(remember to delete the .txt after package.xml).
> [CMakeLists.txt](https://github.com/erik-nelson/blam/files/2507882/CMakeLists.txt)
> [package.xml.txt](https://github.com/erik-nelson/blam/files/2507883/package.xml.txt)
> 
> Another solution is given in the [issue#34](https://github.com/erik-nelson/blam/issues/34#issuecomment-432091700)
Sorry for hijacking the post, but this solved my problem. Thanks.

maggieliuzzi, 2019-08-22:
> Hi, you could replace the two uploaded files in ${your_catkin_ws}/internal/src/geometry_utils(remember to delete the .txt after package.xml). [...]

Finally a solution that works for my case. Thanks!!