erik-nelson/blam #38: Another compilation error : liblaser_loop_closure.so

jattiogbe, 2019-02-11:
Hi,

I am on Ubuntu 18.04, ROS Melodic.
I am trying to compile this project, I used patches described in other issues : 
editing **geometry_utils files** and also **LaserLoopClosure.cc** &  **LaserLoopClosure.h**

Now here is what I get on ./update :

```
[ 88%] Linking CXX shared library /home/julien/BLAM_ws/src/blam/internal/devel/lib/liblaser_loop_closure.so
/usr/bin/ld : ne peut trouver -lBoost::timer
collect2: error: ld returned 1 exit status
laser_loop_closure/CMakeFiles/laser_loop_closure.dir/build.make:961: recipe for target '/home/julien/BLAM_ws/src/blam/internal/devel/lib/liblaser_loop_closure.so' failed
make[2]: *** [/home/julien/BLAM_ws/src/blam/internal/devel/lib/liblaser_loop_closure.so] Error 1
CMakeFiles/Makefile2:4203: recipe for target 'laser_loop_closure/CMakeFiles/laser_loop_closure.dir/all' failed
make[1]: *** [laser_loop_closure/CMakeFiles/laser_loop_closure.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed
```

Any idea ?

Nruano3, 2019-02-12:
So I ended up editing the internal/build/laser_loop_closure/CMakeFiles/laser_loop_closure.dir/link.txt to replace -lboost::timer with the actual library path for boost timer. On my machine that was: /usr/lib/x86_64-linux-gnu/libboost_timer.so

After that I had to remove -lBoostGraphAlgorithms with the path of library for boost_graph. On my machine that was: /usr/lib/x86_64-linux-gnu/libboost_graph.so.

Then I had to also replace all the other -lBoost::Timer from internal/build/blam_slam/CMakeFiles/blam_slam.dir/link.txt, internal/build/blam_slam/CMakeFiles/blam_slam_node.dir/link.txt, and internal/build/blam_slam/CMakeFiles/blam_slam_offline.dir/link.txt

After all these changes everything finished compiling.

jattiogbe, 2019-02-13:
Thanks @Nruano3 for the clear explanation,

I modified everything you wrote except for **lBoostGraphAlgorithms** that I can't find in link.txt, I only have 
**libvtkInfovisBoostGraphAlgorithms**.

I end up with the same error message as before.

closed 2019-02-22 by jattiogbe