erik-nelson/blam #39: Runnning offline error

jattiogbe, 2019-02-22:
Hi ,

When I do roslaunch blam_example test_offline.launch, I get this message :
```

... logging to /home/julien/.ros/log/e14faeb8-3682-11e9-a049-302432a80129/roslaunch-julienBoost-29353.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://julienBoost:44865/

SUMMARY
========

PARAMETERS
 * /blam/blam_slam/check_for_loop_closures: True
 * /blam/blam_slam/filename/bag: /home/julien/BLAM...
 * /blam/blam_slam/filtering/decimate_percentage: 0.9
 * /blam/blam_slam/filtering/grid_filter: True
 * /blam/blam_slam/filtering/grid_res: 0.2
 * /blam/blam_slam/filtering/outlier_filter: False
 * /blam/blam_slam/filtering/outlier_knn: 10
 * /blam/blam_slam/filtering/outlier_std: 1.0
 * /blam/blam_slam/filtering/radius: 0.15
 * /blam/blam_slam/filtering/radius_filter: False
 * /blam/blam_slam/filtering/radius_knn: 3
 * /blam/blam_slam/filtering/random_filter: True
 * /blam/blam_slam/frame_id/base: base
 * /blam/blam_slam/frame_id/fixed: world
 * /blam/blam_slam/frame_id/odometry: odometry
 * /blam/blam_slam/icp/corr_dist: 2.0
 * /blam/blam_slam/icp/iterations: 10
 * /blam/blam_slam/icp/max_rotation: 0.1
 * /blam/blam_slam/icp/max_translation: 0.09
 * /blam/blam_slam/icp/tf_epsilon: 1e-10
 * /blam/blam_slam/icp/transform_thresholding: False
 * /blam/blam_slam/init/orientation/pitch: 0.0
 * /blam/blam_slam/init/orientation/roll: 0.0
 * /blam/blam_slam/init/orientation/yaw: 0.0
 * /blam/blam_slam/init/orientation_sigma/pitch: 0.02
 * /blam/blam_slam/init/orientation_sigma/roll: 0.02
 * /blam/blam_slam/init/orientation_sigma/yaw: 0.02
 * /blam/blam_slam/init/position/x: 0.0
 * /blam/blam_slam/init/position/y: 0.0
 * /blam/blam_slam/init/position/z: 0.0
 * /blam/blam_slam/init/position_sigma/x: 0.1
 * /blam/blam_slam/init/position_sigma/y: 0.1
 * /blam/blam_slam/init/position_sigma/z: 0.1
 * /blam/blam_slam/localization/corr_dist: 1.0
 * /blam/blam_slam/localization/iterations: 10
 * /blam/blam_slam/localization/max_rotation: 0.1
 * /blam/blam_slam/localization/max_translation: 0.05
 * /blam/blam_slam/localization/tf_epsilon: 1e-10
 * /blam/blam_slam/localization/transform_thresholding: False
 * /blam/blam_slam/map/octree_resolution: 0.05
 * /blam/blam_slam/max_tolerable_fitness: 0.15
 * /blam/blam_slam/poses_before_reclosing: 40
 * /blam/blam_slam/proximity_threshold: 1.5
 * /blam/blam_slam/rate/estimate: 20.0
 * /blam/blam_slam/rate/visualization: 0.5
 * /blam/blam_slam/relinearize_skip: 1
 * /blam/blam_slam/relinearize_threshold: 0.01
 * /blam/blam_slam/scan_topic: /velodyne_points
 * /blam/blam_slam/skip_recent_poses: 40
 * /blam/blam_slam/time_end: -1.0
 * /blam/blam_slam/time_scale: -1.0
 * /blam/blam_slam/time_start: 0.0
 * /blam/blam_slam/translation_threshold: 0.5
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /use_sim_time: False

NODES
  /blam/
    blam_slam (blam_slam/blam_slam_offline)

ROS_MASTER_URI=http://localhost:11311

process[blam/blam_slam-1]: started with pid [29369]
[ INFO] [1550839750.581091815]: /blam/blam_slam/BlamSlam: Registering log callbacks.
[ INFO] [1550839750.584959241]: /blam/blam_slam/BlamSlamOffline: Processing the following topics:
	/velodyne_points
[ INFO] [1550839750.598639291]: /blam/blam_slam/BlamSlamOffline: Bag start time    = 1422133388.509928 	 Bag end time    = 1422133476.058435
[ INFO] [1550839750.598669370]: /blam/blam_slam/BlamSlamOffline: Replay start time =         0.000000 	 Replay end time = 4294967296.000000
[ INFO] [1550839750.832518167]: /blam/blam_slam/BlamSlamOffline: Sorting messages.
[ INFO] [1550839750.832770267]: /blam/blam_slam/BlamSlamOffline: Processing messages.
blam_slam_offline: /build/pcl-OilVEB/pcl-1.8.1+dfsg1/octree/include/pcl/octree/impl/octree_pointcloud.hpp:688: void pcl::octree::OctreePointCloud<PointT, LeafContainerT, BranchContainerT, OctreeT>::genOctreeKeyforPoint(const PointT&, pcl::octree::OctreeKey&) const [with PointT = pcl::PointXYZ; LeafContainerT = pcl::octree::OctreeContainerPointIndices; BranchContainerT = pcl::octree::OctreeContainerEmpty; OctreeT = pcl::octree::OctreeBase<pcl::octree::OctreeContainerPointIndices, pcl::octree::OctreeContainerEmpty>]: Assertion `key_arg.x <= this->max_key_.x' failed.
[blam/blam_slam-1] process has died [pid 29369, exit code -6, cmd /home/julien/BLAM_ws/src/blam/internal/devel/lib/blam_slam/blam_slam_offline __name:=blam_slam __log:=/home/julien/.ros/log/e14faeb8-3682-11e9-a049-302432a80129/blam-blam_slam-1.log].
log file: /home/julien/.ros/log/e14faeb8-3682-11e9-a049-302432a80129/blam-blam_slam-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

I couldn't find the default bagfile _lbl_lidar.bag,_ so if you have it somewhere feel free to upload it for me. So I used one called _NSH indoor outdoor.bag_, as described by Erik Nelson [here](https://github.com/erik-nelson/blam/issues/1).

Do you know how I can solve this ?
thanks

AravindChandradoss, 2019-03-17:
Hi, @jattiogbe did you solve this issue? I am also having the same issue.

jattiogbe, 2019-03-17:
Hi,

Sorry I can't help you, no one answered to my issue... Good luck.

AravindChandradoss, 2019-03-17:
uh...? I m trying... did you figure it out? 

-Aravind

AravindChandradoss, 2019-03-17:
https://github.com/erik-nelson/blam/issues/36#issuecomment-462348949

Check this!

stevephone, 2020-09-17:
Do you solve it？
I meet the same problem.

cristianrubioa, 2020-09-27:
I have the same problem.

GabrieleGalimberti-GaleSelector, 2020-11-02:
I have the same problem, 
What should I do?