erik-nelson/blam #40: Make a map

space03, 2019-03-25:
Hi. I'm very newbee in ROS. 
So I don't know about blam and other things....

I recorded rosbag data and i want to make the 3D Map using this data.

How can i do for making 3D map??

For now, 
1st terminal, roslaunch blam_example test_online.launch
2nd terminal, rviz
3rd terminal, rosbag play my_data.bag --clock --topic /velodyne_points

So in Rviz, recorded data is shown but i don't know how to make and save map.

Please help me...

ipinyulun, 2019-03-27:
hi, 
may u share what you been doing?
im stuck at your first point which is roslaunch blam_example test_online.launch
and have no idea what to do next.

what is your command for 2nd and 3rd terminal....
i tried rosrun rviz rviz -f blam_slam
but no luck

MohamedEhabHafez, 2019-04-02:
does the map rotate correctly with the lidar frame on rviz ??

hachbani2019, 2019-05-21:
Any of you did solve the problem ? I'm still stuck at the `roslaunch blam_example test_online.launch`. DOn't understand much aout the difference between the internal and external workspaces and not fiding any documentation or tutorial online...

whuzs, 2020-11-04:
> Hi. I'm very newbee in ROS.
> So I don't know about blam and other things....
>
> I recorded rosbag data and i want to make the 3D Map using this data.
>
> How can i do for making 3D map??
>
> For now,
> 1st [...]

how to make and save map? could you saved?

ipinyulun, 2020-11-05:
> how to make and save map? could you saved?

for velodyne
try https://github.com/koide3/hdl_graph_slam
its easier and works fine