erik-nelson/blam #41: matching the tf frames to the husky frames

MohamedEhabHafez, 2019-04-02:
I am using a veldoyne VLP-16 attached to a husky and it is simulated in gazebo when I launch the blam mapping, I have the map but not rotates correctly with the robot because there is something with the tf frames between the BLAM and the husky.
Anyone can offer a help with setting these frames correctly to match each others??
Any help will be so appreciated.

DangMinh24, 2019-07-11:
Hi, having the same problem at first time. Try running `rosrun rqt_tf_tree rqt_tf_tree` to see the odom name. Then go to `base_frames.yaml` and fix the odometry parameter (ex: `odometry: odom`).

mohit-khosti, 2019-08-29:
Hi @DangMinh24 similarly I can't get a live sensor visualization with my Ouster LiDAR... could you help me somehow?
Thanks

DangMinh24, 2019-08-29:
Hi @mohit-khosti, probably you may want to create a new post for this problem. Anyway, can you check whether there is a `tf` between LIDAR and odometry.

mohit-khosti, 2019-08-29:
Hi @DangMinh24 I did create a post unfortunately no help there and I have tried `rosrun rqt_tf_tree rqt_tf_tree ` and I get "no tf data found" someone suggested I use `static_transform_broadcaster` of `tf2_ros` to broadcast this "base_link -> os1_sensor" transform but I am not sure how to do it since I am a rookie ROS user
Any help would be appreciated!