erik-nelson/blam #42: roslaunch blam_example test_online.launch Operation error

ouyangshihan, 2019-04-20:
[blam/blam_slam-2] process has died [pid 12087, exit code -11, cmd /home/ouyangshihan/BLAM_WS/devel/lib/blam_slam/blam_slam_node ~pcld:=/velodyne_points __name:=blam_slam __log:=/home/ouyangshihan/.ros/log/bcace618-6371-11e9-8b12-00f48d9179bf/blam-blam_slam-2.log].
log file: /home/ouyangshihan/.ros/log/bcace618-6371-11e9-8b12-00f48d9179bf/blam-blam_slam-2*.log

markopuk, 2019-05-21:
same error with test_offline.launch, did you find solution?

thelineover, 2019-05-30:
Can anyone help with this error? please

thelineover, 2019-05-31:
> Can anyone help with this error? please
I solved this Error.
re-compile GTSAM

BMannell, 2019-07-31:
I was getting this error when using the the latest version of the GTSAM develop branch. I was able to resolve it by using the 4.0.0 GTSAM release instead. [https://github.com/borglab/gtsam/releases/tag/4.0.0](https://github.com/borglab/gtsam/releases/tag/4.0.0)

cloitre, 2019-09-06:
I am still having the same error after installing the 4.0.0 version of GTSAM that was linked by @BMannell (Thank you for your help).
I made sure to compile GTSAM from source, and install it in `/usr/local/`
BLAM compiles, but I do get some warnings:

`-- Could NOT find ensenso (missing: ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
** WARNING ** io features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing: DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** io features related to davidSDK will be disabled
-- Could NOT find DSSDK (missing: _DSSDK_LIBRARIES) 
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.`

`CMake Warning at /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'pcl' but neither 'pcl_INCLUDE_DIRS' nor
  'pcl_LIBRARIES' is defined.`

`-- Found Eigen3: /usr/include/eigen3 (Required is at least version "2.91.0") 
CMake Warning at /opt/ros/melodic/share/catkin/cmake/catkin_package.cmake:166 (message):
  catkin_package() DEPENDS on 'Eigen3' but neither 'Eigen3_INCLUDE_DIRS' nor
  'Eigen3_LIBRARIES' is defined.
`

When running the offline launch file, BLAM crashes without more information than this:
`[blam/blam_slam-2] process has died [pid 4236, exit code -11, cmd /home/acloitre/ROS/blam/internal/devel/lib/blam_slam/blam_slam_offline __name:=blam_slam __log:=/home/acloitre/.ros/log/342a44cc-d0e3-11e9-9b5d-0242af234c6a/blam-blam_slam-2.log].
log file: /home/acloitre/.ros/log/342a44cc-d0e3-11e9-9b5d-0242af234c6a/blam-blam_slam-2*.log
'

Any idea of what might be wrong?