erik-nelson/blam #48: Error compiling BLAM: GTSAM built on different Eigen

BMannell, 2019-07-30:
I'm getting a strange error when running ./update.

```
In file included from /home/ben/Documents/blam/internal/src/laser_loop_closure/include/laser_loop_closure/LaserLoopClosure.h:45:0,
                 from /home/ben/Documents/blam/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:37:
/usr/local/include/gtsam/base/Vector.h:72:1: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
 static_assert(
 ^
```

It seems similar to #6. 

Is there a way to enforce either GTSAM or BLAM to use a specific version of Eigen?
GTSAM seems to come packaged with a version of Eigen, is there some way to get it to use the previously installed version?
How can I tell which version is being used in order to align the versions of the other package?

Ubuntu 16.04 LTS
Latest versions of each library from github repos.

BMannell, 2019-07-31:
Solution: Set `GTSAM_USE_SYSTEM_EIGEN=ON` in the GTSAM CMakeCache.txt file.

closed 2019-07-31 by BMannell

Conjuror1279, 2019-11-12:
Hey @BMannell, I am an amateur opencv enthusiast facing the same error as you were earlier, and I also tried setting the GTSAM_USE_SYSTEM_EIGEN=ON in the CMakeCache.txt file, but somehow it did not resolve my error. Could you tell what possibly could be the reason for that and how can I resolve it. Thanks :)

BMannell, 2019-11-15:
@Conjuror1279 
Sorry, I'm afraid I won't be able to help much. I've probably re-installed Ubuntu 3 or 4 times since getting this error and I don't really remember what steps I took during this setup.

Conjuror1279, 2019-11-16:
@BMannell 
        Thank you for your reply, I will see if I can find anything on the internet about the issue.Thanks anyways :).

hilbertletanger, 2021-03-04:
[image]

Note that your "set(GTSAM_USE_SYSTEM_EIGEN ON)" command needs to be placed after "include(cmake/HandleEigen.cmake)". Because in HandleEigen.cmake, GTSAM_USE_SYSTEM_EIGEN will be initialized to off by default.