erik-nelson/blam #50: visualize BLAM with a running LiDAR sensor

mohit-khosti, 2019-08-19:
Hi,
I have installed BLAM and now I am trying to get a live visualization with the help of BLAM, but I do not know how to do it. I used the command `roslaunch blam_example test_online.launch` and then `rviz` but I see nothing. Any help will be appreciated

oldjimmypku, 2019-08-19:
hello there, 
I'm facing the same problem. It's seems like there is no fixed frame for TF. we need to add a tf to rviz,
But I don't know how yet.

mohit-khosti, 2019-08-29:
@oldjimmypku I think it is possible with `static_transform_broadcaster` tho not sure how exactly

DangMinh24, 2019-08-29:
Hi @mohit-khosti , can you check which frame does Ouster LiDAR is using? I have not had experience with Ouster, but I use Velodyne 16 (it uses `/velodyne` frame).

mohit-khosti, 2019-08-29:
Hey @DangMinh24 i believe it is `/os1_cloud_node/points`

DangMinh24, 2019-08-29:
`/os1_cloud_node/points` is a topic, not a `tf` frame. To know which frame LiDAR are using, run the `viz.rviz` file and look at `Global Options -> Fixed Frame`

mohit-khosti, 2019-08-30:
Hi @DangMinh24 thanks for the info, I followed the steps you mentioned and the frames are `os1_sensor` `os1_imu` `os1_lidar` what do I do next?
thanks!

DangMinh24, 2019-08-30:
Hi @mohit-khosti, if you use along with the robot (then there will be a frame called `odom` or `odometry`, then probably you want to make sure there is transition between this frame and the frame that Ouster LIDAR are using (`static_transform_broadcaster` is the right tool). However, if you use alone with LIDAR only, probably you want to change parameters in `test_online.launch` file. 

Try change `/velodyne_points` to `/os1_cloud_node/points`

mohit-khosti, 2019-08-30:
Hi @DangMinh24 I am only working with the LiDAR sensor and no robot. I did change the parameters in `test_online.launch` file, however I see an error in the rviz window "Fixed Frame No tf data. Actual error: Fixed Frame [map] does not exist.
Besides, I am not sure how to implement the `static_transform_broadcaster` solution.
Thank you!

DangMinh24, 2019-08-30:
@mohit-khosti  Can you change `fixed`  parameter back to `world` in `base_frames.yaml` file? If you use sensor alone, the original parameter should work fine.

mohit-khosti, 2019-09-04:
@DangMinh24 I tried looking up the `base_frames.yaml` file in the blam folder, however I cannot find it. Is it possible that you meant `blam_frames.yaml`?
[image]