erik-nelson/blam #55: no matching fucntion to call to 'LaserLoopClosure

Ashik19, 2019-10-30:
Can anyone please help me with this error while installing the blam ?

[image]

Any help will be very appreciated.
Thanks

SKT647, 2021-03-30:
same problem ,how to do

dalexiou48, 2021-06-09:
If you check in BlamSlam.cc file and especially in BlamSlam::HandleLoopClosures method the method AddBetweenFactor has four inputs(it has also an input named stamp). But, in the laser_loop_closure package, the same AddBetweenFactor has 3 inputs. So I suggest to modify this method in laser_loop_closure package and add another dummy input of type same as stamp. It worked for me, in ubuntu 20.04 and ROS Noetic.

qaisarehman, 2021-07-02:
anyone with the fix for this issue! ?